lukasvst/dm-vio #3: Compilation error: Missing gtsam/base/DerivedValue.h

GarronLiu, 2022-03-31:
Firstly, thank you for your contributions! Aimed to fix the problem as shown in the title, I have checked Gtsam file, but got nothing about DerivedValue.h. Could you  give any hints about this? Thank you very much！
[image]

lukasvst, 2022-03-31:
Thanks for checking out the code!
At the moment the newest GTSAM version is not supported yet.
Can you go to the directory where you downloaded GTSAM and run
`git checkout a738529af9754c7a085903f90ae8559bbaa82e75`
and rebuild everything afterwards?

At a later point I will probably upgrade the project to the newest version of GTSAM.

GarronLiu, 2022-04-02:
> Thanks for checking out the code! At the moment the newest GTSAM version is not supported yet. Can you go to the directory where you downloaded GTSAM and run `git checkout [...]

After following your suggestion, the compilation has been successfully done! Thank you again!

closed 2022-04-02 by lukasvst