lukasvst/dm-vio #5: Configuration for EuRoC dataset

hellovuong, 2022-04-02:
Hello, Thank you for the fantastic works. 
I am able to obtain the very precise result in TUM-VI dataset. However, I got struggle on reproduced the result on EuRoC, sequence MH_04 for specific, (also in OpenLoris datasets), the scale seems bad. I generated the config and run by the python scripts. Do you have any suggest to run it correctly?
Thank you in advance!

lukasvst, 2022-04-02:
Thanks for trying out the code!
Could you please send me the folder with the results so I can have a look later today (the name should start with dmvioresult)?
(The folder is generated by the python tools and also contains a file setup/setup.yaml which I can use to see which commands were used to run it.)

hellovuong, 2022-04-02:
Thanks for the quick reply. Here is the [result](https://drive.google.com/file/d/1qiPEdFzZWPDMbw9WmjwHOA9wqfUiOXuj/view?usp=sharing) folder generated by the scripts

lukasvst, 2022-04-02:
Thanks for sending the results!
The configuration looks correct but I'll be only able to evaluate the results in a couple of hours as I'm on my phone right now.

Could you do a full run (with 10 iterations) to see if the difference is up to randomness?
As the method is non-deterministic, individual runs can differ a bit, but in my experience the results on EuRoC are quite stable overall.

hellovuong, 2022-04-02:
Sure, I will run 10 iterations now. But I also think I will not make much difference too :))

hellovuong, 2022-04-15:
Close for inactivity

closed 2022-04-15 by hellovuong