lukasvst/dm-vio #6: problem in building google_tests

JaafarMahmoud, 2022-04-03:
I am trying to build the project, now i am stuck with the following problem.. Can anyone help?

/usr/bin/ld: CMakeFiles/Google_Tests_run.dir/test_PoseTransformationFactor.cpp.o: undefined reference to symbol '_ZN3tbb8internal23allocate_via_handler_v3Em'
/usr/bin/ld: /lib/x86_64-linux-gnu/libtbb.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/Google_Tests_run.dir/build.make:134: test/bin/Google_Tests_run] Error 1
make[1]: *** [CMakeFiles/Makefile2:216: test/CMakeFiles/Google_Tests_run.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: lib/libdmvio.a(BAGTSAMIntegration.cpp.o): undefined reference to symbol '_ZN3tbb8internal23allocate_via_handler_v3Em'
/usr/bin/ld: /lib/x86_64-linux-gnu/libtbb.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dmvio_dataset.dir/build.make:131: bin/dmvio_dataset] Error 1
make[1]: *** [CMakeFiles/Makefile2:160: CMakeFiles/dmvio_dataset.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

lukasvst, 2022-04-03:
It looks like an issue with TBB and GTSAM.
Which operating system are you using and have you installed TBB with `sudo apt install libtbb-dev`?
It might make sense to delete the build folder of DM-VIO and GTSAM and rebuild afterwards.
Also, do the tests and examples from GTSAM work?

JaafarMahmoud, 2022-04-03:
Thank you for your fast reply!
I am building on Ubuntu 20 and yes; I tried installing TBB as you suggested.
Thanks for the suggestion. I will try that.

shreyk25, 2022-04-12:
@lukasvst 
Hi, even I face exactly the same issue.
I am building it on Ubuntu 20.04 and I did install tbb before building it.

There isn't any build folder in dm-vio and gtsam (we create it during building). Could you elaborate on that?

In addition to this, I was curious whether this repo has been tested with more recent gtsam versions? I am guessing they might have resolved this issue in the latest versions.

Thanks!

JaafarMahmoud, 2022-04-12:
@shreyk25 Hello, i tried building on different computer, it was successfully built without any problems... 
I tried so many things on my notebook... But so far couldn't solve it... So reinstalling Ubuntu is an option. 
About building with newer gtsam... I tried that... You will have some issues of using old and different functions from the specified commit...

lukasvst, 2022-04-13:
Thanks for the information on the problem, could you try out the following and let me now if it works?

There is a workaround to make DM-VIO work with the GTSAM version v4.0.3 (which is not the newest one but not as old as the current default).

You need to apply the following patch files to DM-VIO and GTSAM respectively: https://gist.github.com/lukasvst/d2da820f324bac99b4ba0c9dc0bc2a71

In the GTSAM directory run
`git checkout 4.0.3
git apply patch_gtsam_v4_0_3.patch`
and recompile GTSAM with the following cmake command!

```
cmake -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
make -j4
sudo make install
```

In the DM-VIO directory run
`git apply PathDM-VIOForGTSAMv4_0_3.patch`
and recompile.

When running DM-VIO, you might get the error:
`dmvio_dataset: error while loading shared libraries: libmetis-gtsam.so: cannot open shared object file: No such file or directory`
In this case you need to run (and probably recompile)
`export LD_LIBRARY_PATH=/usr/local/lib/:$LD_LIBRARY_PATH` as explained in this issue https://github.com/borglab/gtsam/issues/380

If this helps I will make upgrading GTSAM in the master branch a priority. Otherwise please let me know, then I'll investigate further.

lukasvst, 2022-04-15:
I created a new branch `feature/updated_gtsam` which supports the newest GTSAM version.
To try it out just `git checkout feature/updated_gtsam` and follow the installation instructions for GTSAM again.

If it works I will merge it into master later.

JaafarMahmoud, 2022-04-28:
Hello!
The branch ```feature/updated_gtsam``` worked with latest gtsam with no problems... 
Thanks for your great work.

lukasvst, 2022-05-09:
Thanks for the update, I'm glad it works now.
I have now merged it into master.

closed 2022-05-09 by lukasvst