lukasvst/dm-vio #9: Any plans to release a version that supports ros?

Taeyoung96, 2022-04-14:
@lukasvst  
Hi, Thanks for your great work!  
I have a question about DM-VIO support ROS.  
Do you have any plans to support the ros version?  
Thanks,

lukasvst, 2022-04-14:
It depends on how large the interest in ROS support is and in how much time I will have.
The first plan is to add a live demo (without ROS) for some Realsense cameras, which could be extended to a ROS driver afterwards.

Maybe people who are interested in ROS-support could mark the post with 👍 so I get an overview?

lukasvst, 2022-05-23:
Maybe a small update on this. The live demo for the Realsense T265 is working and the code for this will be published in the next 2-3 weeks. I will also release a ROS driver soon, hopefully next month (June).

Taeyoung96, 2022-05-24:
@lukasvst   
Thank you so much for not forgetting to reply.  
Looking forward to some great code!  :laughing:

lukasvst, 2022-06-15:
The live version for the Realsense T265 is now public, for more information (including some general tips for running on custom datasets) see https://github.com/lukasvst/dm-vio/blob/master/doc/RealsenseLiveVersion.md

The plan is still to release a ROS driver based on this live version towards the end of this month.

lukasvst, 2022-06-22:
The ROS wrapper is now released at https://github.com/lukasvst/dm-vio-ros
I hope it will be useful!

closed 2022-06-22 by lukasvst