lukasvst/dm-vio #10: std::bad_alloc error on NUC with ubuntu 18.04

snakehaihai, 2022-05-04:
HI
I`m encountering issue with the following error. can you help to address this?
```
nuc (master) build $ ./bin/dmvio_dataset files=/home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/images vignette=/home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/vignette.png imuFile=/home/nuc/Desktop/dataset-corridor1_512_16/dso/imu.txt gtFile=/home/nuc/Desktop/dataset-corridor1_512_16/dso/gt_imu.csv calib=/home/nuc/library/dm-vio/configs/tumvi_calib/camera02.txt gamma=/home/nuc/library/dm-vio/configs/tumvi_calib/pcalib.txt imuCalib=/home/nuc/library/dm-vio/configs/tumvi_calib/camchain.yaml mode=0 use16Bit=1 preset=0                                                 # use 1 for realtime nogui=1                                                         # use 1 to enable GUI resultsPrefix=/home/nuc/library/dm-vio/result/result settingsFile=/home/nuc/library/dm-vio/configs/tumvi.yaml start=2
loading data from /home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/images!
loading vignette from /home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/vignette.png!
IMU file is locatd at: /home/nuc/Desktop/dataset-corridor1_512_16/dso/imu.txt!
loading gtFile from /home/nuc/Desktop/dataset-corridor1_512_16/dso/gt_imu.csv!
loading calibration from /home/nuc/library/dm-vio/configs/tumvi_calib/camera02.txt!
loading gammaCalib from /home/nuc/library/dm-vio/configs/tumvi_calib/pcalib.txt!
Loading imu parameters from /home/nuc/library/dm-vio/configs/tumvi_calib/camchain.yaml!
Loading IMU parameter file at: /home/nuc/library/dm-vio/configs/tumvi_calib/camchain.yaml
Used T_cam_imu: 
  -0.999525   0.0296153 -0.00852233   0.0472799
 0.00750192  -0.0343974    -0.99938  -0.0474432
 -0.0298901   -0.998969   0.0341589     -0.0682
          0           0           0           1
Used noise values: 0.00447213 0.0014142 0.316227 0.1
PHOTOMETRIC MODE WITH CALIBRATION!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
Settings:
accelerometer_noise_density: 0.316227
accelerometer_random_walk: 0.00447213
addVisualToCoarseGraphIfTrackingBad: 0
alwaysCanBreakIMU: 0
baToCoarseAccBiasVariance: 1000
baToCoarseGyrBiasVariance: 0.05
baToCoarsePoseVariance: 0.1
baToCoarseRotVariance: 1
baToCoarseVelVariance: 0.1
dynamicWeightRMSEThresh: 8
fixKeyframeDuringCoarseTracking: 1
generalScaleIntervalSize: 60
gravityDirectionFixZ: 1
gyroscope_noise_density: 0.1
gyroscope_random_walk: 0.0014142
init_coarseScaleUncertaintyThresh: 1
init_conversionType: 0
init_disableVIOUntilFirstInit: 1
init_fixPoses: 1
init_initDSOParams: 1
init_lambdaLowerBound: 1e-16
init_maxNumPoses: 100
init_multipleBiases: 0
init_multithreadedInitDespiteNonRT: 0
init_onlyKFs: 1
init_pgba_conversionType: 0
init_pgba_delay: 100
init_pgba_prepareGraphAddDelValues: 0
init_pgba_prepareGraphAddFactors: 0
init_pgba_priorExtrinsicsRot: 0.01
init_pgba_priorExtrinsicsTrans: 0.1
init_pgba_priorGravityDirection: 0.4
init_pgba_priorGravityDirectionZ: 0.0001
init_pgba_reinitScaleUncertaintyThresh: 0.5
init_pgba_scaleUncertaintyThresh: 1
init_pgba_skipFirstKFs: 0
init_priorExtrinsicsRot: 0.01
init_priorExtrinsicsTrans: 0.1
init_priorGravityDirection: 0.4
init_priorGravityDirectionZ: 0.0001
init_priorRotSigma: 1e-05
init_priorTransSigma: 1e-05
init_requestFullResetErrorThreshold: -1
init_requestFullResetNormalizedErrorThreshold: -1
init_scalePriorAfterInit: 0
init_secondthreshGravdir: 1000
init_secondthreshScale: 1e+07
init_threshGravdir: 1000
init_threshScale: 1.02
init_transitionModel: 2
init_updatePoses: 1
integration_sigma: 0.316227
maxFrameEnergyThreshold: 5000
maxPreloadImages: 0
maxTimeBetweenInitFrames: 100000
numMeasurementsGravityInit: 40
preload: 0
priorExtrinsicsRot: 0.01
priorExtrinsicsTrans: 0.1
priorGravityDirection: 0.4
priorGravityDirectionZ: 0.0001
resultsPrefix: 
setting_forceNoKFTranslationThresh: 0
setting_maxOptIterations: 6
setting_minFramesBetweenKeyframes: -0.5
setting_minIdepth: 0.02
setting_minOptIterations: 1
setting_optGravity: 1
setting_optIMUExtrinsics: 0
setting_optScaleBA: 1
setting_prior_bias: 0
setting_prior_velocity: 0
setting_scaleFixTH: 0
setting_solverMode: 2048
setting_transferCovToCoarse: 1
setting_visualOnlyAfterScaleFixing: 0
setting_weightDSOCoarse: 0.001
setting_weightDSOToGTSAM: 1.66667e-05
setting_weightZeroPriorDSOInitX: 0
setting_weightZeroPriorDSOInitY: 0
skipFirstKeyframe: 1
transferCovToCoarseMultiplier: 1
updateDynamicWeightDuringOptimization: 1
use16Bit: 1
useScaleDiagonalHack: 0
Reading Calibration from file /home/nuc/library/dm-vio/configs/tumvi_calib/camera02.txt ... found!
Creating Equidistant undistorter
Input resolution: 512 512
In: EquiDistant 190.978477 190.973307 254.931706 256.897443 0.003482 0.000715 -0.002053 0.000203
Out: 0.200000 0.200000 0.499000 0.499000 0.000000
Output resolution: 512 512

Rectified Kamera Matrix:
  102.4       0 254.988
      0   102.4 254.988
      0       0       1

Reading Photometric Calibration from file /home/nuc/library/dm-vio/configs/tumvi_calib/pcalib.txt
Reading Vignette Image from /home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/vignette.png
Successfully read photometric calibration!
got 5990 images and 5990 timestamps and 5990 exposures.!
ImageFolderReader: got 5990 files in /home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/images!
Skipping comment line in IMU data.
IMU Id: 1520531829221612058
using pyramid levels 0 to 3. coarsest resolution: 64 x 64!
START PANGOLIN!
Using setting_minFramesBetweenKeyframes=0.5 because of non-realtime mode.
Switching to initializer state: CoarseIMUInit
PixelSelector: Using block sizes: 16, 16
Loading gt data
GTData distance (seconds): 0.000900458
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

lukasvst, 2022-05-04:
I think the problem is that the command you're running contains the comments, so everything after the first `#`is not executed anymore.
Can you try the following?

```
./bin/dmvio_dataset files=/home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/images vignette=/home/nuc/Desktop/dataset-corridor1_512_16/dso/cam0/vignette.png imuFile=/home/nuc/Desktop/dataset-corridor1_512_16/dso/imu.txt gtFile=/home/nuc/Desktop/dataset-corridor1_512_16/dso/gt_imu.csv calib=/home/nuc/library/dm-vio/configs/tumvi_calib/camera02.txt gamma=/home/nuc/library/dm-vio/configs/tumvi_calib/pcalib.txt imuCalib=/home/nuc/library/dm-vio/configs/tumvi_calib/camchain.yaml mode=0 use16Bit=1 preset=0 nogui=1 resultsPrefix=/home/nuc/library/dm-vio/result/result settingsFile=/home/nuc/library/dm-vio/configs/tumvi.yaml start=2
```

snakehaihai, 2022-05-09:
Hi I tried and it works. Thanks

May I ask if I have a dataset in ROS bag format. How do i convert it into your reqired format? is there script that I can use?

snakehaihai, 2022-05-09:
ok thx. i`ll try

closed 2022-05-09 by snakehaihai