lukasvst/dm-vio #12: Program runs without result

HappyB1rthday, 2022-05-11:
[image]
Hi, The program is running, but there is no graphic or dot on Pangolin. The dataset is TUM-VI slides1. The resultsPrefix file exists but without anything. Is that normal? Or is there something wrong with the format of my input?
Thanks for your help.

lukasvst, 2022-05-11:
No, this is not normal, you should see the images in the bottom left of the Pangolin window and the system should make it past the initialization phase.
The command looks ok at first glance.

Can you please post the full commandline output (especially the part before what's shown in the screenshot) so I can check if there is anything wrong with it?

HappyB1rthday, 2022-05-11:
[image]
[image]
Could you please check it for me

lukasvst, 2022-05-11:
I don't see anything wrong in the output before it not initializing. I think there is something wrong with the installation, but it is a bit tricky to debug from a distance.
* Can you try to add the argument `useimu=0` and see if this changes something?
* Have you changed anything in the code? Maybe you could try the newest commit of the master branch.
* Another thing to try is to change line 66 in the file `src/dso/FullSystem/CoarseInitializer.cpp` to `printDebug=true;` 
This will enable more debug output. Could you paste the output when using this setting (maybe as a code block instead of a screenshot, as it will be quite long)?

lukasvst, 2022-05-11:
One more thing: Could you try to set `mode=1` in the commandline arguments? As the images don't seem to arrive, the input to the photometric calibration could be wrong and this will disable it.

lukasvst, 2022-06-15:
Do you have an update on this, were you able to resolve the issue? Otherwise I will close the issue in a week.

For future reference: My best guess is that the problem was that the images were somehow loaded as 8bit instead of 16bit. In this case a different pcalib file would need to be passed, otherwise the behavior is similar to what the @HappyB1rthday reported. For quickly testing this it is best to try to set `mode=1`as this mode does not use the pcalib file.

closed 2022-06-24 by lukasvst

GeJintian, 2023-09-18:
> Do you have an update on this, were you able to resolve the issue? Otherwise I will close the issue in a week.
>
> For future reference: My best guess is that the problem was that the images were [...]

I meet exactly the same problem, and setting mode=1 solves this problem. Thanks a lot.

lukasvst, 2023-09-23:
Thanks for letting me know! 

One thing to note is that `mode=1` disables the photometric calibration which will slightly reduce the accuracy. (But it's great to try for debugging.)

So you could try using `mode=0` and passing `gamma=PATH_TO_DMVIO/configs/pcalib_linear_8bit.txt` which should hopefully work a bit better.

The explanation for this is that 8 bit images need a different pcalib (camera response calibration) file than 16 bit images, and in your case it seems like the images are loaded as 8bit.