lukasvst/dm-vio #13: Pangolin X11: Failed to open X display

shreyk25, 2022-05-14:
Hi, on running run_dmvio.py on the euroc dataset, I get the error saying 
Pangolin X11: Failed to open X display,

Could you please let me know how I can resolve this?

lukasvst, 2022-05-16:
Sometimes this error can be printed but the program will still work fine. 
Does the code work without GUI (just pass `nogui=1`)?
If not, can you post the command you run and the full commandline output?

shreyk25, 2022-05-24:
Thanks for pointing that out. Looks like the problem is not with Pangolin X11.
I get the following error:
`terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to open X display
Aborted (core dumped)
Executing: cp /home/shreyas/dm_vio_test/results/dmvioresult-euroc-2022-05-24--17-40-16/mav_V2_03_difficult_0/result.txt /home/shreyas/dm_vio_test/results/dmviore
sult-euroc-2022-05-24--17-40-16/results/mav_V2_03_difficult_0.txt               
cp: cannot stat '/home/shreyas/dm_vio_test/results/dmvioresult-euroc-2022-05-24--17-40-16/mav_V2_03_difficult_0/result.txt': No such file or directory           
Executing: cp /home/shreyas/dm_vio_test/results/dmvioresult-euroc-2022-05-24--17-40-16/mav_V2_03_difficult_0/resultKFs.txt /home/shreyas/dm_vio_test/results/dmvi
oresult-euroc-2022-05-24--17-40-16/kfres/mav_V2_03_difficult_0.txt              
cp: cannot stat '/home/shreyas/dm_vio_test/results/dmvioresult-euroc-2022-05-24--17-40-16/mav_V2_03_difficult_0/resultKFs.txt': No such file or directory`

This error says there is no results.txt file. However, isn't it supposed to be created by the code. Have I messed up any configuration settings? Your help will be really appreciated.

lukasvst, 2022-06-27:
Do you have any update on this, otherwise I will close the issue in a couple of days.

closed 2022-07-04 by shreyk25