lukasvst/dm-vio #14: run dm-vio with realsense d435i

jy84-choi, 2022-05-16:
To run dm-vio with realsense d435i data,
First, realsense d435i data was saved in EuRoC format.
I also made camera.txt, imu.txt, and times.txt files
And I ran dm-vio with the files.
I'm getting the following nan error:

`
813-stats: 0
814-stats: 0
LOG 191: -nan fine. Res: 0 A, 0 L, 0 M; (0 / 0) forceDrop. a=0.000000, b=0.000000. Window 191 (2)
GTData distance (seconds): 0.00414728
New ref frame id: 191 prepared keyframe id: 191
Prepared keyframe id: 191
lvl3, it -1 (l=0.010000 / 1.000000) INITIA: 0.000->-nan (0 -> 0) (|inc| = 0.000000)! 	0 0 0 0 0 0 AFF 0 0 (rel 1 0)
lvl 3, it 0 (l=0.010000 / 1.000000) REJECT: -nan->-nan (0 -> 0) (|inc| = -nan)! 	0 0 0 0 0 0 AFF 0 0 (rel 1 0)
inc too small, break!
WARNING: Coarse tracker thinks that tracking was not good!
BIG ERROR! tracking failed entirely. Take predicted pose and hope we may somehow recover.
Predicted pose:
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
Coarse Tracker tracked ab = 0.000000 0.000000 (exp 1.000000). Res nan!
Time since last keyframe: 0.0333569
Frame history size: 193
Preparing keyframe: 192
Frames between KFs: 0
SPARSITY:  MinActDist 0.600000 (need 1000 points, have 0 points)!
OPTIMIZE 0 pts, 0 active res, 0 lin res!
activeResiduals.size(): 0
lastEnergy: 0
0
0
lastEnergyL: 0
lastEnergyM: nan
Initial Error       	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab 0.000000 0.000000!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 0 (L -5.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 1 (L -3.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 2 (L -1.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 3 (L 1.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 4 (L 3.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 5 (L 5.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 6 (L 7.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 7 (L 9.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 8 (L 11.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 9 (L 13.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 10 (L 15.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 11 (L 17.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 12 (L 19.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 13 (L 21.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Dynamic weight: 1
STEPS: A -nan; B -nan; R nan; T -nan. 	REJECT 14 (L 23.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab -nan -nan!
Num BA Iterations done: 15
LOG 192: -nan fine. Res: 0 A, 0 L, 0 M; (0 / 0) forceDrop. a=0.000000, b=0.000000. Window 192 (3)
GTPose: -0.023315 -0.050298         0
[... 149 lines cut ...]
inc too small, break!
WARNING: Coarse tracker thinks that tracking was not good!
RE-TRACK ATTEMPT 29 with initOption 29 and start-lvl 3 (ab 0.000000 0.000000): nan nan nan nan nan -> nan nan nan -nan nan 
lvl3, it -1 (l=0.010000 / 1.000000) INITIA: 0.000->-nan (0 -> 0) (|inc| = 0.000000)! 	-nan -nan  nan -nan -nan -nan AFF 0 0 (rel 1 0)
lvl 3, it 0 (l=0.010000 / 1.000000) REJECT: -nan->-nan (0 -> 0) (|inc| = -nan)! 	-nan -nan  nan -nan -nan -nan AFF 0 0 (rel 1 0)
inc too small, break!
WARNING: Coarse tracker thinks that tracking was not good!
RE-TRACK ATTEMPT 30 with initOption 30 and start-lvl 3 (ab 0.000000 0.000000): nan nan nan nan nan -> nan nan nan -nan nan 
BIG ERROR! tracking failed entirely. Take predicted pose and hope we may somehow recover.
Predicted pose:
-nan -nan -nan  nan
-nan -nan -nan -nan
-nan -nan -nan -nan
   0    0    0    1
TRACKING FAILED ENTIRELY, NO HOPE TO RECOVER
TRACKING FAILED ENTIRELY, NO HOPE TO RECOVER
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  14 (X_GetGeometry)
  Resource id in failed request:  0x4400003
  Serial number of failed request:  3278
  Current serial number in output stream:  3278
`

When I took a log in the code, I checked that all residuals are in ResState::OOB.
How can I solve this?

lukasvst, 2022-05-16:
Thanks for first trying the noimu mode first.
Have you also passed the camchain file and is it read correctly?

Could you please post the command you use for running and the beginning of the commandline output where all the settings are printed, until the error first occurs?

jy84-choi, 2022-05-17:
Thank you for your reply!

I made camchain file using the default camera-inertial calibration values of realsense d435i.

My run command is below:
python3 run_dmvio.py --output=console --dataset=sr --dmvio_settings=sr.yaml --withgui --iter=1 --only_seq=1 --dmvio_args="init_pgba_skipFirstKFs=1 init_requestFullResetNormalizedErrorThreshold=0.8"

Here, sr is my dataset name.

Since "#-stats: 0" repeats a lot, I deleted "#-stats:" log for readability.

----------- create_dmvio_commands -----------
files=./data calib=./camera.txt imuCalib=/home/jychoi/git_ws/src/dm-vio/configs/sr_calib/camchain.yaml mode=1 preset=0 nogui=0 useimu=1 quiet=0 settingsFile=/home/jychoi/git_ws/src/dm-vio/configs/sr.yaml init_pgba_skipFirstKFs=1 init_requestFullResetNormalizedErrorThreshold=0.8
WARNING: No end times for dataset.
----------- Save Project Status -----------
{'name': 'dmvioresult', 'dataset': 'sr', 'build_type': 'RelWithDebInfo', 'num_iter': 1, 'only_seq': 1, 'results_name': 'dmvioresult-sr-2022-05-17--09-57-11', 'config_name': 'workpc', 'date_run': datetime.datetime(2022, 5, 17, 9, 57, 11, 382691), 'realtime': False, 'temporary': False, 'noimu': False, 'quiet': False, 'output_type': 'console', 'withgui': True, 'custom_dmvio_args': 'init_pgba_skipFirstKFs=1 init_requestFullResetNormalizedErrorThreshold=0.8', 'dmvio_settings': 'sr.yaml', 'gdb': False, 'git_hash': '5bf2a67c10973a91c061483b10a0257e2e0c5b11', 'commit_message': 'Added note on the GTSAM version to the README.', 'commit_time': datetime.datetime(2022, 5, 4, 15, 26, 58), 'diff_empty': False, 'eval_tool_command': 'run_dmvio.py --output=console --dataset=sr --dmvio_settings=sr.yaml --withgui --iter=1 --only_seq=1 --dmvio_args=init_pgba_skipFirstKFs=1 init_requestFullResetNormalizedErrorThreshold=0.8', 'eval_tools_git_hash': 'f1e581e26022706b97aceee8022489e8a97e5c82', 'eval_tools_commit_message': 'Initial commit.', 'eval_tools_diff_empty': False, 'command_0': '/home/jychoi/git_ws/src/dm-vio/cmake-build-relwithdebinfo/bin/dmvio_dataset files=./data calib=./camera.txt imuCalib=/home/jychoi/git_ws/src/dm-vio/configs/sr_calib/camchain.yaml mode=1 preset=0 nogui=0 useimu=1 quiet=0 settingsFile=/home/jychoi/git_ws/src/dm-vio/configs/sr.yaml init_pgba_skipFirstKFs=1 init_requestFullResetNormalizedErrorThreshold=0.8 resultsPrefix=/home/jychoi/git_ws/src/dm-vio-result/dmvioresult-sr-2022-05-17--09-57-11/sr_220426_162017_cam_40_light_on_ir_color_2_0/ start=1', 'cwd0': '/home/jychoi/dataset/SR_VIO_EuRoC/220426_162017_cam_40_light_on_ir_color_2/cam0'}

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

----------- STARTING EXECUTION! -----------
Working Dir: /home/jychoi/dataset/SR_VIO_EuRoC/220426_162017_cam_40_light_on_ir_color_2/cam0
Command: /home/jychoi/git_ws/src/dm-vio/cmake-build-relwithdebinfo/bin/dmvio_dataset files=./data calib=./camera.txt imuCalib=/home/jychoi/git_ws/src/dm-vio/configs/sr_calib/camchain.yaml mode=1 preset=0 nogui=0 useimu=1 quiet=0 settingsFile=/home/jychoi/git_ws/src/dm-vio/configs/sr.yaml init_pgba_skipFirstKFs=1 init_requestFullResetNormalizedErrorThreshold=0.8 resultsPrefix=/home/jychoi/git_ws/src/dm-vio-result/dmvioresult-sr-2022-05-17--09-57-11/sr_220426_162017_cam_40_light_on_ir_color_2_0/ start=1
loading data from ./data!
loading calibration from ./camera.txt!
Loading imu parameters from /home/jychoi/git_ws/src/dm-vio/configs/sr_calib/camchain.yaml!
Loading IMU parameter file at: /home/jychoi/git_ws/src/dm-vio/configs/sr_calib/camchain.yaml
Used T_cam_imu: 
      1       0       0   0.006
      0       1       0 -0.0051
      0       0       1  -0.012
      0       0       0       1
Used noise values: 0.00447213 0.0014142 0.316227 0.1
PHOTOMETRIC MODE WITHOUT CALIBRATION!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
Enabling IMU integration!
Loading settings from yaml file: /home/jychoi/git_ws/src/dm-vio/configs/sr_calib/camchain.yaml!
START AT 1!
Settings:
accelerometer_noise_density: 0.2
accelerometer_random_walk: 0.04
addVisualToCoarseGraphIfTrackingBad: 0
alwaysCanBreakIMU: 0
baToCoarseAccBiasVariance: 1000
baToCoarseGyrBiasVariance: 0.05
baToCoarsePoseVariance: 0.1
baToCoarseRotVariance: 1
baToCoarseVelVariance: 0.1
dynamicWeightRMSEThresh: 8
fixKeyframeDuringCoarseTracking: 1
generalScaleIntervalSize: 60
gravityDirectionFixZ: 1
gyroscope_noise_density: 0.02
gyroscope_random_walk: 0.004
init_coarseScaleUncertaintyThresh: 1
init_conversionType: 0
init_disableVIOUntilFirstInit: 1
init_fixPoses: 1
init_initDSOParams: 1
init_lambdaLowerBound: 1e-16
init_maxNumPoses: 100
init_multipleBiases: 0
init_multithreadedInitDespiteNonRT: 0
[... 90 lines cut ...]
START PANGOLIN!
Using setting_minFramesBetweenKeyframes=0.5 because of non-realtime mode.
Switching to initializer state: CoarseIMUInit
PixelSelector: Using block sizes: 16, 16
Loading gt data
GTData distance (seconds): 0.0078123
PixelSelector: Using block sizes: 16, 16
GTData distance (seconds): 0.0067623
InitTimeBetweenFrames: 0.0333583

# GTdata distance and InitTimeBetweenFrames are repeated.

Scaling with rescaleFactor: 1.28913
Initialization: keep 10.9% (need 1000, have 9202)!
INITIALIZE FROM INITIALIZER (974 pts)!
Frame history size: 192
Preparing keyframe: 191
Frames between KFs: 190
SPARSITY:  MinActDist 1.900000 (need 1000 points, have 974 points)!
OPTIMIZE 974 pts, 974 active res, 0 lin res!
activeResiduals.size(): 974
Initial Error       	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab 0.000000 0.000000!
Dynamic weight: 1
STEPS: A nan; B nan; R nan; T nan. 	REJECT 0 (L -5.00, dir nan, ss 1.0): 	A(0.000000)=(AV -nan). Num: A(0) + M(0); ab nan nan!
Dynamic weight: 1

lukasvst, 2022-05-17:
The problem is that the IMU data cannot be associated with the image data.
The line `IMU-data too old -> skipping frame` should only printed a couple of times maximum. If it is printed out a lot and with the line `Found no start frame for IMU-data!` there is no frame with associated IMU data.

Have you interpolated the IMU file, so that each frame has an IMU measurement with exactly its timestamp (e.g. with [this script](https://github.com/lukasvst/dm-vio-python-tools/blob/master/interpolate_imu_file.py)?

If that's not the issue, the association code makes some strong assumptions:
* The first column in `times.txt` must be the timestamp of the corresponding image in nanoseconds. There should be a measurement in the IMU file with exactly the same timestamp. 
* Images are associated with the line in `times.txt` simply by order. It's best practice to name each image after the first column in times.txt (the timestamp in nanoseconds), but this is not checked at runtime.

I hope this helps in finding the issue, (otherwise you could also post your `times.txt` and `imu.txt`file.)

jy84-choi, 2022-05-17:
When realsense d435i is turned on, imu and camera are not completely synchronized, but I thought that they bring data of a somewhat similar time.
If you look at the time.txt and imu.txt files below, you can see that they have data at the same time although there is a slight difference.

======= times.txt ==========
1652235648154954240 1652235648.154954240
1652235648188310272 1652235648.188310272
1652235648221666304 1652235648.221666304
1652235648255022336 1652235648.255022336
1652235648288378368 1652235648.288378368
1652235648321734400 1652235648.321734400
1652235648355090432 1652235648.355090432
1652235648388446208 1652235648.388446208
1652235648421802240 1652235648.421802240
1652235648455158272 1652235648.455158272
1652235648488514048 1652235648.488514048
1652235648521871104 1652235648.521871104
1652235648555227904 1652235648.555227904
1652235648588584960 1652235648.588584960
1652235648621942016 1652235648.621942016
1652235648655298816 1652235648.655298816
1652235648688655872 1652235648.688655872
1652235648722012672 1652235648.722012672
1652235648755369728 1652235648.755369728
1652235648788726528 1652235648.788726528
1652235648822083584 1652235648.822083584
1652235648855440384 1652235648.855440384
1652235648888797440 1652235648.888797440
1652235648922154240 1652235648.922154240
1652235648955511296 1652235648.955511296
1652235648988868096 1652235648.988868096
1652235649022225152 1652235649.022225152
1652235649055581696 1652235649.055581696
1652235649088938752 1652235649.088938752
1652235649122295552 1652235649.122295552

======= imu.txt ==========
1652235648124645888 -0.001701 0.001742 0.000013 0.095851 -7.562911 6.369858
1652235648129643264 -0.001701 0.013959 0.000013 0.095851 -7.562911 6.369858
1652235648134639616 -0.003447 -0.008730 0.000013 0.095851 -7.562911 6.369858
1652235648139636224 0.000044 -0.003494 -0.001733 0.095851 -7.562911 6.369858
1652235648144632576 -0.001701 0.001742 0.000013 0.095943 -7.542913 6.370518
1652235648149628928 -0.003447 0.003487 -0.001733 0.095943 -7.542913 6.370518
1652235648154626560 0.000044 -0.000003 0.000013 0.095943 -7.542913 6.370518
1652235648159622912 0.000044 -0.000003 0.000013 0.095943 -7.542913 6.370518
1652235648164619264 0.000044 -0.001749 0.000013 0.095943 -7.542913 6.370518
1652235648169615360 0.000044 -0.008730 0.000013 0.095943 -7.542913 6.370518
1652235648174612992 -0.001701 -0.000003 0.000013 0.095851 -7.562911 6.369858
1652235648179609344 -0.001701 -0.001749 -0.001733 0.095851 -7.562911 6.369858
1652235648184605696 0.000044 -0.000003 0.000013 0.095851 -7.562911 6.369858
1652235648189602304 -0.003447 0.001742 0.000013 0.115927 -7.563198 6.349864
1652235648194598656 -0.003447 0.001742 -0.001733 0.115927 -7.563198 6.349864
1652235648199595008 -0.003447 0.008723 -0.001733 0.115927 -7.563198 6.349864
1652235648204592384 -0.003447 -0.000003 0.000013 0.055941 -7.542293 6.371044
1652235648209588992 0.000044 -0.000003 0.000013 0.055941 -7.542293 6.371044
1652235648214585088 -0.003447 -0.000003 -0.001733 0.055941 -7.542293 6.371044
1652235648219581440 -0.003447 -0.005239 0.000013 0.075850 -7.562601 6.370121
1652235648224579072 0.000044 0.005233 0.000013 0.075850 -7.562601 6.370121
1652235648229575424 -0.003447 0.003487 -0.003478 0.075850 -7.562601 6.370121
1652235648234571776 0.000044 0.017450 0.000013 0.075850 -7.562601 6.370121
1652235648239568128 0.000044 -0.000003 0.000013 0.095851 -7.562911 6.369858
1652235648244564736 -0.003447 -0.000003 -0.001733 0.095851 -7.562911 6.369858
1652235648249562112 -0.005192 -0.000003 0.000013 0.095851 -7.562911 6.369858
1652235648254558464 -0.003447 -0.000003 -0.001733 0.075867 -7.542627 6.390511
1652235648259554560 0.000044 -0.000003 0.000013 0.075867 -7.542627 6.390511
1652235648264551168 0.000044 0.003487 -0.001733 0.075867 -7.542627 6.390511
1652235648269547520 -0.003447 -0.000003 -0.001733 0.095851 -7.562911 6.369858

lukasvst, 2022-05-18:
There are two "levels" of IMU-camera synchronization:
* The first one is that IMU and camera timestamps are recorded with the same device or otherwise made consistent. This is a prerequisite for running DM-VIO, and I think for the D435i this is the case.
* The second level is that the IMU is triggered manually and always records an IMU sample exactly during the timestamp (middle of exposure) of each image. E.g. the VI-Sensor used in the EuRoC dataset does this, but most other visual-inertial sensors don't. For these sensors you can still run DM-VIO, but first you need to add a "fake IMU measurement" for each camera timestamp, by interpolating the neighboring IMU samples. As mentioned, [this script](https://github.com/lukasvst/dm-vio-python-tools/blob/master/interpolate_imu_file.py) does this for you.

So long story short: You need to run
`python3 interpolate_imu_file --input imu.txt --times times.txt --output pass_this_imu_file_to_dmvio.txt` and then it should work.

Please let me know if this solves the issue for you.

jy84-choi, 2022-05-20:
> There are two "levels" of IMU-camera synchronization:
>
> * The first one is that IMU and camera timestamps are recorded with the same device or otherwise made consistent. This is a prerequisite for [...]

It works!!!!!!!!!
Thank you so much for being patient with me. 
I think it will be of great help to others if it is included in the readme.

closed 2022-05-20 by jy84-choi

lukasvst, 2022-05-20:
I'm glad that it helped. Yes, I will include it in the README with a later update.