lukasvst/dm-vio #21: Live streaming video to dm-vio from a raspberry pi camera

DyumanAditya, 2022-07-09:
Hi,
I've built dm-vio on a raspberry pi, and I want to be able to stream video from the raspberry pi camera module to dm-vio. Is this possible? I can use OpenCV to capture the images from the live stream, so it's just a question of passing image frames to dm-vio. How can I do this?
Thanks

WFram, 2022-07-09:
Hi, you can use a ROS node for Raspberry Pi camera module (https://github.com/dganbold/raspicam_node), which lets you publish your image data in ROS message format. In order to subscribe to them, use DM-VIO ROS wrapper (https://github.com/lukasvst/dm-vio-ros).

DyumanAditya, 2022-07-09:
Thanks for your answer. I'm trying to avoid using ROS, is there another way of doing this? I have access to each of the images coming from the livestream by using OpenCV, how can I feed these images into dm-vio?

lukasvst, 2022-07-09:
You can write your own executable, similar to the ROS wrapper or the RealsenseLiveDemo: https://github.com/lukasvst/dm-vio/blob/master/src/main_dmvio_t265.cpp
You just need to feed the images and IMU data into the IMUInterpolator, see https://github.com/lukasvst/dm-vio/blob/master/src/live/RealsenseT265.cpp#L151 https://github.com/lukasvst/dm-vio/blob/master/src/live/RealsenseT265.cpp#L169 and https://github.com/lukasvst/dm-vio/blob/master/src/live/RealsenseT265.cpp#L211