lukasvst/dm-vio #23: If we can work well when visual info is not good at a short time

yushijie94, 2022-08-08:
I want to know, if I go into a black house, so I can not get any visual info. Can It still work by using IMU info at a short time and when I get out the black house, I can get the true trajectory showing I was in and out.
Thanks!

lukasvst, 2022-08-09:
It depends on how little visual information there is and on the time.
When there is absolutely no visual information, only the IMU can be used for pose estimation and this will drift very quickly (few seconds). Note, that this is a pure odometry system, so it will usually not correct the trajectory after regaining vision.

But if you still have some visual information, or loose it only for a short period of time, the system should work.

yushijie94, 2022-08-09:
> It depends on how little visual information there is and on the time. When there is absolutely no visual information, only the IMU can be used for pose estimation and this will drift very quickly [...]

OK, If I test in real data, I will try to verify the conclusion. Thanks for you reply!

closed 2022-08-09 by yushijie94

Leerw, 2022-10-28:
@yushijie94 hi, are there any conclusions?