lukasvst/dm-vio #24: tbb problem

Liu2022, 2022-08-10:
Hello,everyone.

I have followed the readme.md, but meet the problem below when I comfile dm-vio.
/usr/bin/ld: lib/libdmvio.a(IMUIntegration.cpp.o): undefined reference to symbol '_ZN3tbb6detail2r117deallocate_memoryEPv'
/usr/bin/ld: /usr/local/lib/libtbb.so.12: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

It seems to be the former issue https://github.com/lukasvst/dm-vio/issues/6.

But I have tried the same way (as below) to install my gtsam but the problem still exists.

cmake -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
make -j4
sudo make install

My gtsam version is 4.2a6. And my tbb version is 2021.5.0.

I hope someone can give me a hand. Thank you very much!

lukasvst, 2022-08-10:
It seems like GTSAM does not propagate tbb as a linking dependency.
Which operating system are you using?
Is anyone else having the issue, I couldn't reproduce it on my machine unfortunately?

One workaround is to just compile GTSAM without tbb (this might have a slight impact on runtime, but I think not too serious)
`cmake -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DGTSAM_WITH_TBB=OFF ..`

You could also try out this: https://github.com/borglab/gtsam/issues/240#issuecomment-637670334

Lastly, you could explicitly link DM-VIO against tbb and tbbmalloc (by adding them to https://github.com/lukasvst/dm-vio/blob/master/CMakeLists.txt#L192 )

Please let me know which of these solutions work for you.

Liu2022, 2022-08-11:
Thank you for your reply.
It seems that my tbb is too high. Last night I changed my tbb from 2021.5.0 to 2020.1 and recompiled the gtsam and DM-VIO.
Eventually everything worked well.
Thank you for your advices! 
My operating system is ubuntu 20.04.

lukasvst, 2022-08-16:
Thanks for letting us know the solution!
I'll close it for now, maybe later I will have time to fix it for newer tbb or include it as a requirement in the documentation.

closed 2022-08-16 by lukasvst