lukasvst/dm-vio #26: SSE2NEON.h not found

bot-lin, 2022-09-18:
I am building the project in Docker on a M2 silicon chip Mac. I followed the installation instruction and I got the following error when run make -j4
`root@8ae9bd6c5112:/dm-vio/build# make -j1
Scanning dependencies of target dmvio
[  1%] Building CXX object CMakeFiles/dmvio.dir/src/dso/FullSystem/FullSystem.cpp.o
In file included from /dm-vio/src/dso/FullSystem/CoarseTracker.h:34,
                 from /dm-vio/src/dso/FullSystem/FullSystem.cpp:49:
/dm-vio/src/dso/OptimizationBackend/MatrixAccumulators.h:29:10: fatal error: SSE2NEON.h: No such file or directory
   29 | #include "SSE2NEON.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/dmvio.dir/build.make:63: CMakeFiles/dmvio.dir/src/dso/FullSystem/FullSystem.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:125: CMakeFiles/dmvio.dir/all] Error 2
make: *** [Makefile:84: all] Error 2`

bot-lin, 2022-09-18:
i forgot this install  sse2neon.

closed 2022-09-18 by bot-lin

setareh-soltanieh, 2024-08-09:
Hi, were you able to create the Docker files for this project? @bot-lin