lukasvst/dm-vio #28: cannot find -lBoost::regex

shervinkoushan, 2022-09-24:
This happens during the last build step, even though `libboost` is installed.

For reference DSO built just fine.

shervinkoushan, 2022-09-24:
Solved (I believe the actual error was caused by yaml-cpp)

closed 2022-09-24 by shervinkoushan

LvDingzhou, 2022-12-20:
hi,I met the same problem,can you tell more details about the solution?appreciate!

LvDingzhou, 2022-12-21:
> Are running Ubuntu 18.04? Can you try compiling with a newer version of cmake and let me know if that helps?

yep,I'm running Ubuntu 18.04 on Jetson.
I 've runned DSO successfully on this device,and meet errors when cmake dm-vio as follows:

>CMake Error at CMakeLists.txt:191 (add_executable):  Target "dmvio_dataset" links to target "Boost::regex" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED target, or an ALIAS target is missing?

> CMake Error at test/CMakeLists.txt:7 (add_executable): Target "Google_Tests_run" links to target "Boost::regex" but the target was  not found.  Perhaps a find_package() call is missing for an IMPORTED target, or an ALIAS target is missing?

The cmake version is 3.10.2-1ubuntu2.18.04.2  .I will try compiling with a newer version of cmake .

LvDingzhou, 2022-12-21:
> Are running Ubuntu 18.04? Can you try compiling with a newer version of cmake and let me know if that helps?

It works,I compiled dm-vio with cmake 3.24.3,at least,no error for "Boost::regex"any more.
now,I'm going to change boost version.

> CMake Error at /usr/local/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
>   Could NOT find Boost: Found unsuitable version "1.60.0", but required is at
>   least "1.65" (found /usr/local/include, found components: serialization
>   system filesystem thread program_options date_time timer chrono regex
>   atomic)
> Call Stack (most recent call first):
>   /usr/local/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:592 (_FPHSA_FAILURE_MESSAGE)
>   /usr/local/share/cmake-3.24/Modules/FindBoost.cmake:2376 (find_package_handle_standard_args)
>   /usr/local/share/cmake-3.24/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
>   /usr/local/lib/cmake/GTSAM/GTSAMConfig.cmake:21 (find_dependency)
>   CMakeLists.txt:20 (find_package)

LvDingzhou, 2022-12-22:
> It works,I compiled dm-vio with cmake 3.24.3,at least,no error for "Boost::regex"any more. now,I'm going to change boost version.

After changing the boost version,I runned DM-VIO successfully,thanks for your advice about cmake ,it helps,Lukas!
Ubuntu18.04 Jetson,cmake 3.24.3 (cmake 3.10.1, installed by JetPack,brings problems),boost 1.65.1.0ubuntu1

lukasvst, 2022-12-22:
I'm glad to hear that it works now!