lukasvst/dm-vio #30: Running for more than 5 minutes is prone to the question about the number of gradient points extracted is abnormally small

seanXMU, 2022-11-11:
normal situation:
[image]
abnormal situation:
[image]

I'm sorry to bother you. Could you tell me why this situation happened？

lukasvst, 2022-11-18:
This looks like something that would happen when the camera pose diverges or gets very inaccurate. How does the trajectory look like up to this point?

Does it always happen after more than 5 minutes, or just when certain things happen (e.g. strong accelerations, standing still etc.)?

If the system gets instable after some time it might have something to do with the IMU biases, which parameters are you using for the bias random walk (accelerometer_random_walk and gyroscope_random_walk), maybe they are too small?

Can you try to run the system without IMU (`useimu=0`) and tell me if the issue also arises there?

Count-C, 2023-01-25:
i have the same situation both in kitti and m2dgr dataset，and i tried to set useimu=0 on kitti and it works well.
how can i solve it?

lukasvst, 2023-01-28:
On Kitti, it is expected that the visual-inertial version will not work. The reason is that the IMU data on Kitti is not well time-synchronized and as far as I know there is also some missing IMU data in some sequences.

Regarding m2dgr, I am not sure as this dataset should have good time-synchronization and calibration (although I have not tried the dataset yet). Are you sure that you provided the correct calibration and are using the IMU which is on the VI-sensor? It might be a good idea to increase the IMU noise parameters which you pass to DM-VIO.

Count-C, 2023-02-13:
Thanks for reply!

In M2DGR, I changed the IMU to the VI-sensor and increase the noise and random walk 10 times. It works .

I wander if the dm-vio can only use the 6-axis IMU? Because the 9-axis IMU offered in M2DGR did not work at all.  And I also tested it on another dataset VECtor Benchmark (https://star-datasets.github.io/vector/) which offers 9-axis IMU, and it did not work either.

seanXMU, 2023-02-19:
I'm sorry to reply to you so late. I found that if I run the system without IMU (useimu=0), this problem would disappear. 
I think it may be that the parameters of IMU  are not good enough

closed 2023-02-20 by seanXMU