lukasvst/dm-vio #31: Some cmake error

HaisongW, 2022-11-18:
Hi,

I followed the instruction of README to build dm-vio, but the cmake generate step failed. The cmake error information is as follows:

```
haisongwang@haisongwang-ThinkPad-Edge-E540:~/SLAM/dm-vio/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system thread filesystem chrono serialization date_time timer regex 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found LIBZIP: /usr/lib/x86_64-linux-gnu/libzip.so  
--- found PANGOLIN, compiling with pangolin library.
--- found OpenCV, compiling with opencv library.
--- found ziplib (1.5.1), compiling with zip capability.
--- compiling dmvio_dataset.
-- Configuring done
CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

That's wired, because Eigen3 is found, and Eigen3::Eigen is not included in DOVIO_LINKED_LIBRARIES, so I can't fix it by modifying CMakeLists.txt file.
Before build dm-vio, I have build LDSO successfully.

Any suggestions? Thanks very much.

Forgive my poor English...

HaisongW, 2022-11-18:
That's the original cmake generate error information.
` haisongwang@haisongwang-ThinkPad-Edge-E540:~/SLAM/dm-vio/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system thread filesystem chrono serialization date_time timer 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found LIBZIP: /usr/lib/x86_64-linux-gnu/libzip.so  
--- found PANGOLIN, compiling with pangolin library.
--- found OpenCV, compiling with opencv library.
--- found ziplib (1.5.1), compiling with zip capability.
--- compiling dmvio_dataset.
-- Configuring done
CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
`

lukasvst, 2022-11-18:
Which operating system are you using, and which versions of cmake and Eigen3?

If LDSO compiles and DM-VIO doesn't, it might have something to do with GTSAM. Did the GTSAM build (with DGTSAM_USE_SYSTEM_EIGEN=ON) work successfully and are you using version 4.2a6?

Sometimes it can help to use a newer cmake version.

HaisongW, 2022-11-21:
> 
Hi, Lukasvst, thanks for your reply.My operating system is Ubuntu-20.04, cmake version is 3.16.3, Eigen3 version is 3.4.90.

I am not very sure if the GTSAM was build with DGTSAM_USE_SYSTEM_EIGEN=ON, so I removed all files in /usr/local/include/gtsam and /usr/local/include/gtsam, then re-cmake and re-make gtsam, the gtsam cmake information is as follows:

```
(base) haisongwang@haisongwang-ThinkPad-Edge-E540:~/SLAM/gtsam/build$ cmake -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM Version: 4.2a6
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Found Eigen version: 3.4.90
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.2.0
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.2.0
-- Install prefix: /usr/local
-- Wrote /home/haisongwang/SLAM/gtsam/build/GTSAMConfig.cmake
-- Wrote /home/haisongwang/SLAM/gtsam/build/GTSAM_UNSTABLEConfig.cmake
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen dot 
[... 84 lines cut ...]
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system thread filesystem chrono serialization date_time timer 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found LIBZIP: /usr/lib/x86_64-linux-gnu/libzip.so  
--- found PANGOLIN, compiling with pangolin library.
--- found OpenCV, compiling with opencv library.
--- found ziplib (1.5.1), compiling with zip capability.
--- compiling dmvio_dataset.
-- Configuring done
CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```
I dont understand...

HaisongW, 2022-11-21:
> Which operating system are you using, and which versions of cmake and Eigen3?
>
> If LDSO compiles and DM-VIO doesn't, it might have something to do with GTSAM. Did the GTSAM build (with [...]

I got something new, after updating cmake version from 3.16.3 to the newest 3.25.0, the cmake information of dm-vio changed:
```
haisongwang@haisongwang-ThinkPad-Edge-E540:~/SLAM/dm-vio/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system thread filesystem chrono serialization date_time timer 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
CMake Warning (dev) at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIBZIP)
  does not match the name of the calling package (LibZip).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindLibZip.cmake:24 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found LIBZIP: /usr/lib/x86_64-linux-gnu/libzip.so  
--- found PANGOLIN, compiling with pangolin library.
--- found OpenCV, compiling with opencv library.
--- found ziplib (1.5.1), compiling with zip capability.
--- compiling dmvio_dataset.
-- Configuring done
CMake Error at /usr/local/lib/cmake/Pangolin/PangolinTargets.cmake:74 (set_target_properties):
  The link interface of target "pango_geometry" contains:

    Eigen3::Eigen

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/local/lib/cmake/Pangolin/PangolinConfig.cmake:6 (include)
  CMakeLists.txt:33 (find_package)

CMake Error at /usr/local/lib/cmake/Pangolin/PangolinTargets.cmake:99 (set_target_properties):
  The link interface of target "pango_opengl" contains:

    Eigen3::Eigen

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/local/lib/cmake/Pangolin/PangolinConfig.cmake:6 (include)
  CMakeLists.txt:33 (find_package)

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

it seems something wrong with Pangolin, I will try to fix it...

HaisongW, 2022-11-22:
After turning Pangolin version to v0.6, dm-vio build successfully, thanks you Lucasvst, I will close this issue.

closed 2022-11-22 by HaisongW