lukasvst/dm-vio #38: Cmake Error

yuchenwengwyc, 2023-02-21:
I got a cmake error as:

CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:191 (add_executable):
  Target "dmvio_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

when I try to recmake it, it shows that: Pangolin is not linked, this bug is so wired. I don't think that Pangolin is related to Eigen3.  

I have also tried to add an Eigen3 path to Cmakelists.txt, but it still fails and shows a same error. And my Eigen version is 3.3.4.

lukasvst, 2023-02-27:
Are you using Pangolin v0.6 as recommended?

This might be a duplicate of https://github.com/lukasvst/dm-vio/issues/31
If the Pangolin version is not the problem you can also try using a newer cmake version and see if it provides a better error message.

leonidk, 2023-02-27:
I had this issue because GTSAM updated how they handle Eigen3 (even in 4.2a9). I could fix this either using 4.2a6 or replacing the dm-vio FindEigen.cmake in cmake/ with the HandleEigen.cmake (from GTSAM's cmake/).