lukasvst/dm-vio #40: Nothing happened on the GUI

GarronLiu, 2023-03-24:
These are my running command with configurations and information printed on the terminal. And I can't find anything happened on the GUI.Could you please help me?
./bin/dmvio_dataset files=/home/garronliu/data/E/bagfiles/dataset1/dso/cam0/images vignette=/home/garronliu/data/E/bagfiles/dataset1/dso/cam0/vignette.png imuFile=/home/garronliu/data/E/bagfiles/dataset1/dso/imu.txt gtFile=/home/garronliu/data/E/bagfiles/dataset1/dso/gt_imu.csv calib=/home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/camera02.txt gamma=/home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/pcalib.txt  imuCalib=/home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/camchain.yaml mode=0 use16Bit=1 preset=1 nogui=0 resultsPrefix=/home/garronliu/dmvio_results settingsFile=/home/garronliu/SLAM/dm-vio-master/configs/tumvi.yaml
PHOTOMETRIC MODE WITH CALIBRATION!

=============== PRESET Settings: ===============
DEFAULT settings:
- 1x real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
Loading settings from yaml file: /home/garronliu/SLAM/dm-vio-master/configs/tumvi.yaml!
Loading IMU parameter file at: /home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/camchain.yaml
Used T_cam_imu: 
  -0.999525   0.0296153 -0.00852233   0.0472799
 0.00750192  -0.0343974    -0.99938  -0.0474432
 -0.0298901   -0.998969   0.0341589     -0.0682
          0           0           0           1
Used noise values: 0.043 0.0011 0.224 0.0128
Settings:
accelerometer_noise_density: 0.224
accelerometer_random_walk: 0.043
addVisualToCoarseGraphIfTrackingBad: 0
alwaysCanBreakIMU: 0
baToCoarseAccBiasVariance: 1000
baToCoarseGyrBiasVariance: 0.05
baToCoarsePoseVariance: 0.1
baToCoarseRotVariance: 1
baToCoarseVelVariance: 0.1
calib: /home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/camera02.txt
dynamicWeightRMSEThresh: 8
end: 100000
files: /home/garronliu/data/E/bagfiles/dataset1/dso/cam0/images
fixKeyframeDuringCoarseTracking: 1
gamma: /home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/pcalib.txt
generalScaleIntervalSize: 60
gravityDirectionFixZ: 1
gtFile: /home/garronliu/data/E/bagfiles/dataset1/dso/gt_imu.csv
gyroscope_noise_density: 0.0128
gyroscope_random_walk: 0.0011
imuCalib: /home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/camchain.yaml
imuFile: /home/garronliu/data/E/bagfiles/dataset1/dso/imu.txt
init_coarseScaleUncertaintyThresh: 1
init_conversionType: 0
init_disableVIOUntilFirstInit: 1
init_fixPoses: 1
init_initDSOParams: 1
init_lambdaLowerBound: 1e-16
init_maxNumPoses: 100
init_multipleBiases: 0
init_multithreadedInitDespiteNonRT: 0
init_onlyKFs: 1
init_pgba_conversionType: 0
init_pgba_delay: 100
init_pgba_prepareGraphAddDelValues: 0
init_pgba_prepareGraphAddFactors: 0
init_pgba_priorExtrinsicsRot: 0.01
init_pgba_priorExtrinsicsTrans: 0.1
init_pgba_priorGravityDirection: 0.4
init_pgba_priorGravityDirectionZ: 0.0001
init_pgba_reinitScaleUncertaintyThresh: 0.5
init_pgba_scaleUncertaintyThresh: 1
init_pgba_skipFirstKFs: 0
init_priorExtrinsicsRot: 0.01
init_priorExtrinsicsTrans: 0.1
init_priorGravityDirection: 0.4
init_priorGravityDirectionZ: 0.0001
init_priorRotSigma: 1e-05
init_priorTransSigma: 1e-05
init_requestFullResetErrorThreshold: -1
init_requestFullResetNormalizedErrorThreshold: -1
init_scalePriorAfterInit: 0
init_secondthreshGravdir: 1000
init_secondthreshScale: 1e+07
init_threshGravdir: 1000
init_threshScale: 1.02
init_transitionModel: 2
init_updatePoses: 1
integration_sigma: 0.316227
maxFrameEnergyThreshold: 5000
maxPreloadImages: 0
maxTimeBetweenInitFrames: 100000
numMeasurementsGravityInit: 40
preload: 1
priorExtrinsicsRot: 0.01
priorExtrinsicsTrans: 0.1
priorGravityDirection: 0.4
priorGravityDirectionZ: 0.0001
resultsPrefix: /home/garronliu/dmvio_results
reverse: 0
sampleoutput: 0
setting_forceNoKFTranslationThresh: 0
setting_maxOptIterations: 6
setting_minFramesBetweenKeyframes: -0.5
setting_minIdepth: 0.02
setting_minOptIterations: 1
setting_optGravity: 1
setting_optIMUExtrinsics: 0
setting_optScaleBA: 1
setting_prior_bias: 0
setting_prior_velocity: 0
setting_scaleFixTH: 0
setting_solverMode: 2048
setting_transferCovToCoarse: 1
setting_visualOnlyAfterScaleFixing: 0
setting_weightDSOCoarse: 0.001
setting_weightDSOToGTSAM: 1.66667e-05
setting_weightZeroPriorDSOInitX: 0
setting_weightZeroPriorDSOInitY: 0
skipFirstKeyframe: 0
speed: 1
start: 0
transferCovToCoarseMultiplier: 1
updateDynamicWeightDuringOptimization: 1
use16Bit: 1
useScaleDiagonalHack: 0
vignette: /home/garronliu/data/E/bagfiles/dataset1/dso/cam0/vignette.png
Reading Calibration from file /home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/camera02.txt ... found!
Creating Equidistant undistorter
Input resolution: 512 512
In: EquiDistant 190.978477 190.973307 254.931706 256.897443 0.003482 0.000715 -0.002053 0.000203
Out: 0.200000 0.200000 0.499000 0.499000 0.000000
Output resolution: 512 512

Rectified Kamera Matrix:
  102.4       0 254.988
      0   102.4 254.988
      0       0       1

Reading Photometric Calibration from file /home/garronliu/SLAM/dm-vio-master/configs/tumvi_calib/pcalib.txt
Reading Vignette Image from /home/garronliu/data/E/bagfiles/dataset1/dso/cam0/vignette.png
Successfully read photometric calibration!
got 16213 images and 16213 timestamps and 16213 exposures.!
ImageFolderReader: got 16213 files in /home/garronliu/data/E/bagfiles/dataset1/dso/cam0/images!
Skipping comment line in IMU data.
IMU Id: 1520603863154984026
using pyramid levels 0 to 3. coarsest resolution: 64 x 64!
START PANGOLIN!
Switching to initializer state: RealtimeCoarseIMUInitState
PixelSelector: Using block sizes: 16, 16
LOADING ALL IMAGES!
OpenGL Error 502: GL_INVALID_OPERATION: The specified operation is not allowed in the current state.
In: /usr/local/include/pangolin/gl/gl.hpp, line 214
^CCaught signal 2
segment fault(core dumped)

[image]

GarronLiu, 2023-03-24:
In case that the problem are related to the compilation， the compilation process is shown below：

 make -j4
[  1%] Building CXX object CMakeFiles/dmvio.dir/src/dso/FullSystem/FullSystem.cpp.o
[  3%] Building CXX object CMakeFiles/dmvio.dir/src/dso/FullSystem/FullSystemOptimize.cpp.o
[  5%] Building CXX object CMakeFiles/dmvio.dir/src/dso/FullSystem/FullSystemOptPoint.cpp.o
[  7%] Building CXX object CMakeFiles/dmvio.dir/src/dso/FullSystem/FullSystemDebugStuff.cpp.o
In file included from /home/garronliu/SLAM/dm-vio-master/src/IMU/IMUIntegration.hpp:49:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:44,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.cpp:34:
/home/garronliu/SLAM/dm-vio-master/src/util/GTData.hpp:43:50: warning: ‘SE3’ is deprecated [-Wdeprecated-declarations]
                   Eigen::Vector3d biasTranslation)
                                                  ^
In file included from /home/garronliu/SLAM/dm-vio-master/src/dso/util/NumType.h:29:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:31,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.cpp:34:
/home/garronliu/SLAM/dm-vio-master/thirdparty/Sophus/sophus/se3.hpp:612:7: note: declared here
 class SE3Group : public SE3GroupBase<SE3Group<_Scalar,_Options> > {
       ^~~~~~~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/IMU/IMUIntegration.hpp:49:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:44,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.cpp:34:
/home/garronliu/SLAM/dm-vio-master/src/util/GTData.hpp:48:17: warning: ‘Sophus::SE3’ is deprecated [-Wdeprecated-declarations]
     Sophus::SE3 pose;
                 ^~~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/dso/util/NumType.h:29:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:31,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.cpp:34:
/home/garronliu/SLAM/dm-vio-master/thirdparty/Sophus/sophus/se3.hpp:35:26: note: declared here
 typedef SE3Group<double> SE3 EIGEN_DEPRECATED;
                          ^~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/IMU/IMUIntegration.hpp:49:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:44,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemDebugStuff.cpp:34:
/home/garronliu/SLAM/dm-vio-master/src/util/GTData.hpp:43:50: warning: ‘SE3’ is deprecated [-Wdeprecated-declarations]
                   Eigen::Vector3d biasTranslation)
                                                  ^
In file included from /home/garronliu/SLAM/dm-vio-master/src/dso/util/NumType.h:29:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:31,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemDebugStuff.cpp:34:
/home/garronliu/SLAM/dm-vio-master/thirdparty/Sophus/sophus/se3.hpp:612:7: note: declared here
 class SE3Group : public SE3GroupBase<SE3Group<_Scalar,_Options> > {
       ^~~~~~~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/IMU/IMUIntegration.hpp:49:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:44,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemDebugStuff.cpp:34:
/home/garronliu/SLAM/dm-vio-master/src/util/GTData.hpp:48:17: warning: ‘Sophus::SE3’ is deprecated [-Wdeprecated-declarations]
     Sophus::SE3 pose;
                 ^~~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/dso/util/NumType.h:29:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:31,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemDebugStuff.cpp:34:
/home/garronliu/SLAM/dm-vio-master/thirdparty/Sophus/sophus/se3.hpp:35:26: note: declared here
 typedef SE3Group<double> SE3 EIGEN_DEPRECATED;
                          ^~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/IMU/IMUIntegration.hpp:49:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:44,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemOptimize.cpp:28:
/home/garronliu/SLAM/dm-vio-master/src/util/GTData.hpp:43:50: warning: ‘SE3’ is deprecated [-Wdeprecated-declarations]
                   Eigen::Vector3d biasTranslation)
                                                  ^
In file included from /home/garronliu/SLAM/dm-vio-master/src/dso/util/NumType.h:29:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:31,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemOptimize.cpp:28:
/home/garronliu/SLAM/dm-vio-master/thirdparty/Sophus/sophus/se3.hpp:612:7: note: declared here
 class SE3Group : public SE3GroupBase<SE3Group<_Scalar,_Options> > {
       ^~~~~~~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/IMU/IMUIntegration.hpp:49:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:44,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystemOptimize.cpp:28:
[... 2313 lines cut ...]
[... cut ...]
/home/garronliu/SLAM/dm-vio-master/src/dso/FullSystem/FullSystem.h:275:22: warning: ‘Sophus::SE3’ is deprecated [-Wdeprecated-declarations]
  std::vector<Sophus::SE3> gtPoses;
                      ^~~
In file included from /home/garronliu/SLAM/dm-vio-master/src/dso/util/NumType.h:29:0,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/OptimizationBackend/EnergyFunctionalStructs.h:28,
                 from /home/garronliu/SLAM/dm-vio-master/src/dso/OptimizationBackend/EnergyFunctionalStruc

lukasvst, 2023-03-24:
Thanks for providing all the relevant information.
* You need to add `start=2` to the command, then it should work as intended (sometimes in datasets the image data starts before IMU data, for that reason it's generally advised to skip the first 1-2 images).
* Note that you are using the realtime version which will first load all images into RAM. That's why it might take a while until you see something happening. You can switch to the non-realtime version by changing to `preset=0`, or you can disable this behaviour by setting `preload=0`
* You might also want to append a trailing `/`to your `resultsPrefix`, otherwise it will save the results in the folder above.

I hope this helps!

GarronLiu, 2023-03-25:
> Thanks for providing all the relevant information.
>
>     * You need to add `start=2` to the command, then it should work as intended (sometimes in datasets the image data starts before IMU data, for [...]

Thank you for kind suggestions! Again, thank you for great contributions! Your work is impressive! Now I have already succeeded in running your code in ROS version on my own device in real time! The performance is fantastic! I plan to implement your SLAM code on a small USV to evaluate its performance in inland waterway!

closed 2023-03-25 by GarronLiu