lukasvst/dm-vio #43: Unresolved symbols

antithing, 2023-07-04:
Hi, and thank you for making this code available. I am porting it to Windows, and have successfully compiled the lib, however I see some 'unresolved external symbol' errors when I compile the T265 demo, related to the code in the repo.

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2001	unresolved external symbol "public: int __cdecl dso::PixelSelector::makeMaps(struct dso::FrameHessian const * const,float *,float,int,bool,float)" (?makeMaps@PixelSelector@dso@@QEAAHQEBUFrameHessian@2@PEAMMH_NM@Z)	dmvio	D:\Testing\DM-VIO\dm-vio-master\build\CoarseInitializer.obj	1	

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "public: int __cdecl dso::PixelSelector::makeMaps(struct dso::FrameHessian const * const,float *,float,int,bool,float)" (?makeMaps@PixelSelector@dso@@QEAAHQEBUFrameHessian@2@PEAMMH_NM@Z) referenced in function "private: void __cdecl dso::FullSystem::makeNewTraces(struct dso::FrameHessian *,float *)" (?makeNewTraces@FullSystem@dso@@AEAAXPEAUFrameHessian@2@PEAM@Z)	dmvio	D:\Testing\DM-VIO\dm-vio-master\build\FullSystem.obj	1	

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "public: void __cdecl dso::CoarseTracker::setCoarseTrackingRef(class std::vector<struct dso::FrameHessian *,class std::allocator<struct dso::FrameHessian *> >)" (?setCoarseTrackingRef@CoarseTracker@dso@@QEAAXV?$vector@PEAUFrameHessian@dso@@V?$allocator@PEAUFrameHessian@dso@@@std@@@std@@@Z) referenced in function "private: void __cdecl dso::FullSystem::makeKeyFrame(struct dso::FrameHessian *)" (?makeKeyFrame@FullSystem@dso@@AEAAXPEAUFrameHessian@2@@Z)	dmvio	D:\Testing\DM-VIO\dm-vio-master\build\FullSystem.obj	1	

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "public: void __cdecl dso::CoarseDistanceMap::makeDistanceMap(class std::vector<struct dso::FrameHessian *,class std::allocator<struct dso::FrameHessian *> >,struct dso::FrameHessian *)" (?makeDistanceMap@CoarseDistanceMap@dso@@QEAAXV?$vector@PEAUFrameHessian@dso@@V?$allocator@PEAUFrameHessian@dso@@@std@@@std@@PEAUFrameHessian@2@@Z) referenced in function "private: void __cdecl dso::FullSystem::activatePointsMT(void)" (?activatePointsMT@FullSystem@dso@@AEAAXXZ)	dmvio	D:\Testing\DM-VIO\dm-vio-master\build\FullSystem.obj	1	

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "char const * __cdecl pangolin::glErrorString(unsigned int)" (?glErrorString@pangolin@@YAPEBDI@Z) referenced in function "public: __cdecl pangolin::GlTexture::GlTexture(int,int,int,bool,int,unsigned int,unsigned int,void *)" (??0GlTexture@pangolin@@QEAA@HHH_NHIIPEAX@Z)	dmvio	D:\Testing\DM-VIO\dm-vio-master\build\PangolinDSOViewer.obj	1	
```

The last one is Pangolin, but I am confused about the others, do you have any thoughts on what I might be missing here?

Thank you!

lukasvst, 2023-07-07:
Does the version on datasets (e.g. on TUM-VI corridor 1) work properly or does it also crash?

It looks like it tries to save the calibration and results in the root directory of your drive, I'm not sure if it has write access on Windows for this directory.

Otherwise the best way to debug this error would be to run it with the debugger and check the stacktrace when the error happens.

antithing, 2023-07-11:
Hi @lukasvst thank you for getting back to me. Running in Debug gives me:

```
NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 4. coarsest resolution: 53 x 50!
Using factory IMU calibration!
START PANGOLIN!
Switching to initializer state: RealtimeCoarseIMUInitState
'rm' is not recognized as an internal or external command,
operable program or batch file.
A subdirectory or file logs already exists.
'rm' is not recognized as an internal or external command,
operable program or batch file.
A subdirectory or file mats already exists.
PixelSelector: Using block sizes: 16, 16
SKIPPING 4 FRAMES! frames remaining in queue: 0
PixelSelector: Using block sizes: 16, 16
SKIPPING 0 FRAMES! frames remaining in queue: 4
In: D:\thirdParty\Pangolin-master\include\pangolin/gl/gl.hpp, line 214
InitTimeBetweenFrames: 0.0333335
SKIPPING 0 FRAMES! frames remaining in queue: 5
InitTimeBetweenFrames: 0.0666668
SKIPPING 0 FRAMES! frames remaining in queue: 6
InitTimeBetweenFrames: 0.1
SKIPPING 0 FRAMES! frames remaining in queue: 8
InitTimeBetweenFrames: 0.133334
SKIPPING 0 FRAMES! frames remaining in queue: 9
InitTimeBetweenFrames: 0.166667
SKIPPING 0 FRAMES! frames remaining in queue: 11
InitTimeBetweenFrames: 0.200001
SKIPPING 0 FRAMES! frames remaining in queue: 13
InitTimeBetweenFrames: 0.233334
SKIPPING 0 FRAMES! frames remaining in queue: 15
InitTimeBetweenFrames: 0.266667
SKIPPING 0 FRAMES! frames remaining in queue: 16
InitTimeBetweenFrames: 0.300001
SKIPPING 0 FRAMES! frames remaining in queue: 17
InitTimeBetweenFrames: 0.333334
SKIPPING 0 FRAMES! frames remaining in queue: 21
InitTimeBetweenFrames: 0.366668
SKIPPING 0 FRAMES! frames remaining in queue: 23
InitTimeBetweenFrames: 0.400001
SKIPPING 0 FRAMES! frames remaining in queue: 24
InitTimeBetweenFrames: 0.433334
SKIPPING 0 FRAMES! frames remaining in queue: 26
InitTimeBetweenFrames: 0.466668
SKIPPING 0 FRAMES! frames remaining in queue: 28
InitTimeBetweenFrames: 0.500001
SKIPPING 0 FRAMES! frames remaining in queue: 30
InitTimeBetweenFrames: 0.533334
SKIPPING 0 FRAMES! frames remaining in queue: 31
InitTimeBetweenFrames: 0.566668
SKIPPING 0 FRAMES! frames remaining in queue: 32
InitTimeBetweenFrames: 0.600001
SKIPPING 0 FRAMES! frames remaining in queue: 35
InitTimeBetweenFrames: 0.633334
SKIPPING 0 FRAMES! frames remaining in queue: 38
InitTimeBetweenFrames: 0.666668
SKIPPING 0 FRAMES! frames remaining in queue: 38
InitTimeBetweenFrames: 0.700001
SKIPPING 0 FRAMES! frames remaining in queue: 39
InitTimeBetweenFrames: 0.733335
SKIPPING 0 FRAMES! frames remaining in queue: 42
InitTimeBetweenFrames: 0.766668
SKIPPING 0 FRAMES! frames remaining in queue: 45
InitTimeBetweenFrames: 0.800001
SKIPPING 0 FRAMES! frames remaining in queue: 46
InitTimeBetweenFrames: 0.833335
[... 142 lines cut ...]

It seems to run, and prints this:

```
Current mapping id: 795 create KF after: 597
SKIPPING 1 FRAMES! frames remaining in queue: 1124
lvl4, it -1 (l=0.010000 / 1.000000) INITIA: 0.000->52.568 (0 -> 293) (|inc| = 0.000000)!          0.00285573 -1.31823e-05   0.00260694   0.00733675    0.0349193  -0.00731295 AFF -0.0696036   -21.3145 (rel 0.932796    2.496)
lvl 4, it 0 (l=0.010000 / 1.000000) ACCEPT: 52.568->45.632 (293 -> 293) (|inc| = 0.010617)!     -0.00535996  0.00270789  0.00516661  0.00694484   0.0385329 -0.00598132 AFF -0.0870911   -17.6893 (rel 0.916626  5.70842)
lvl 4, it 1 (l=0.005000 / 1.000000) ACCEPT: 45.632->45.447 (293 -> 295) (|inc| = 0.002700)!     -0.00551905  0.00298596   0.0075905   0.0066162    0.038631  -0.0055161 AFF -0.0936622   -16.9519 (rel 0.910622  6.29255)
lvl 4, it 2 (l=0.002500 / 1.000000) ACCEPT: 45.447->44.646 (295 -> 299) (|inc| = 0.004170)!     -0.00880916  0.00414628  0.00959426  0.00624058   0.0396082 -0.00530238 AFF -0.0955588   -16.8275 (rel 0.908897  6.37298)
lvl 4, it 3 (l=0.001250 / 1.000000) ACCEPT: 44.646->44.455 (299 -> 299) (|inc| = 0.002194)!      -0.0107272  0.00500734
 0.00990702  0.00614703   0.0399294 -0.00511979 AFF -0.0932735   -17.1362 (rel 0.910976  6.11728)
lvl 4, it 4 (l=0.000625 / 1.124683) ACCEPT: 44.455->44.248 (299 -> 299) (|inc| = 0.001463)!      -0.0113372  0.00488776   0.0111308  0.00604992   0.0403728 -0.00530828 AFF -0.0922948   -17.2196 (rel 0.911868  6.05666)
lvl 4, it 5 (l=0.000312 / 1.337481) ACCEPT: 44.248->44.057 (299 -> 299) (|inc| = 0.002398)!      -0.0128475  0.00559371   0.0127964  0.00594494   0.0408012 -0.00528112 AFF -0.0925453   -17.1953 (rel 0.91164 6.07514)
lvl 4, it 6 (l=0.000156 / 1.590541) REJECT: 44.057->44.137 (299 -> 299) (|inc| = 0.002032)!      -0.0116611  0.00456846   0.0116854  0.00549699   0.0406654 -0.00555329 AFF -0.0952043   -16.9333 (rel 0.909219  6.27537)
lvl 4, it 7 (l=0.001000 / 1.000000) REJECT: 44.057->44.063 (299 -> 299) (|inc| = 0.001272)!      -0.0121042  0.00495188   0.0121016  0.00566451   0.0407167 -0.00545228 AFF -0.0942036   -17.0321 (rel 0.910129   6.1998)
lvl 4, it 8 (l=0.004000 / 1.000000) REJECT: 44.057->44.063 (299 -> 299) (|inc| = 0.001252)!      -0.0121135  0.00496145   0.0121143  0.00566873   0.0407198 -0.00545237 AFF -0.0941578   -17.0373 (rel 0.910171  6.19566)
lvl 4, it 9 (l=0.016000 / 1.000000) REJECT: 44.057->44.064 (299 -> 299) (|inc| = 0.001181)!      -0.0121473  0.00499659   0.0121611  0.00568454   0.0407311 -0.00545194 AFF -0.0939995   -17.0554 (rel 0.910315  6.18127)
lvl 4, it 10 (l=0.064000 / 1.000000) REJECT: 44.057->44.068 (299 -> 299) (|inc| = 0.000971)!     -0.0122487  0.00510065   0.0122994  0.00573444   0.0407614 -0.00544348 AFF -0.0936067    -17.101 (rel 0.910673  6.14478)
inc too small, break!
```

Is this a normal output?

Thank you!

lukasvst, 2023-07-13:
The output with `disableAllDisplay = turn;` looks normal, if it's running like that for some time (until the dataset is finished).

Thanks for debugging that the error lies in the GUI. 
Unfortunately I'm not sure how much I will be able to help you with it, as I haven't run DM-VIO on Windows yet and don't plan to officially support it for now.
But maybe you can get some information from this PR: https://github.com/lukasvst/dm-vio/pull/29

antithing, 2023-07-24:
Thank you @lukasvst ! Using that PR has a TUM dataset (dataset-corridor1_512_16) running on Windows. However it is running very slowly (around 10Hz) on a high spec desktop machine. 

I see that there is an option to use a realtime preset :

`preset=0                                                        # use 1 for realtime`

But when i do that, i see:

```
Skipping comment line in IMU data.
IMU Id: 1520531829221612058
using pyramid levels 0 to 3. coarsest resolution: 64 x 64!
START PANGOLIN!
Switching to initializer state: RealtimeCoarseIMUInitState
PixelSelector: Using block sizes: 16, 16
LOADING ALL IMAGES!
OpenGL Error 502: GL_INVALID_OPERATION: The specified operation is not allowed in the current state.
In: D:\Testing\DM-VIO\dm-vio\thirdparty\Pangolin\include\pangolin/gl/gl.hpp, line 214
```

What can i do to solve this?

Thank you

antithing, 2023-07-24:
More on speed.... Using:
`
preset=2 nogui=1 `

Runs, and gives me:

```
5987 Frames (20.0 fps)
65.91ms per frame (single core);
infms per frame (multi core);
0.759x (single core);
0.000x (multi core);
```

Is that an expected speed? I am looking to hit 50 fps if I can.

Thanks!

lukasvst, 2023-07-24:
Good to hear that is running with that PR, but it should definitely be running a lot faster. On a high-end machine I would expect at least 40fps (if your disk loads the images fast enough).

Can you upload the `timings.txt` here, which is generated in the results folder?
In general, running it through a profiler might help to figure out why it's slow.

As to presets: 
* preset=0 will try to run as fast as possible so it's the right one to test your speed with.
* preset=1 is simulating real-time execution (meaning it will not run slower nor faster than 20fps on TUM-VI), and it will also use 2 different threads for front-end tracking and back-end bundle adjustment, making slightly more random. Note that it will first load all images into RAM before proceeding, unless you add `preload=0`
* presets 2 and 3 are legacy and not supported.

antithing, 2023-07-25:
Thank you! I have run the sequnece using:

`dmvio_dataset.exe files=D:/dataSet/dataset-corridor1_512_16/dso/cam0/images vignette=D:/dataSet/dataset-corridor1_512_16/dso/cam0/vignette.png  imuFile=D:/dataSet/dataset-corridor1_512_16/dso/imu.txt gtFile=D:/dataSet/dataset-corridor1_512_16/dso/gt_imu.csv calib=D:/Testing/DM-VIO/dm-vio/configs/tumvi_calib/camera02.txt gamma=D:/Testing/DM-VIO/dm-vio/configs/tumvi_calib/pcalib.txt imuCalib=D:/Testing/DM-VIO/dm-vio/configs/tumvi_calib/camchain.yaml mode=0 use16Bit=1 preset=0 nogui=0 resultsPrefix=results/ settingsFile=D:/Testing/DM-VIO/dm-vio/configs/tumvi.yaml start=2 useimu=1`

and I see:

```
======================
5987 Frames (20.0 fps)
126.60ms per frame (single core);
infms per frame (multi core);
0.395x (single core);
0.000x (multi core);
======================
[timings.txt](https://github.com/lukasvst/dm-vio/files/12159663/timings.txt)
[usedSettingsdso.txt](https://github.com/lukasvst/dm-vio/files/12159664/usedSettingsdso.txt)

```

Timings data attached. Thanks!

lukasvst, 2023-07-25:
Thanks for sending the timings.
It seems to be quite slow overall (i.e. not only in a particular part of the code).
For reference: On my machine addActiveFrame takes 0.021s on average compared to your 0.11s and fullCoarseTracking takes 0.0028s, where your results show 0.068s.

Are you sure that you are building in `RelWithDebInfo` mode (and not in Debug)?