lukasvst/dm-vio #46: How do I compare the data in the ground truth and the data in the result?

GY56723-debug, 2023-07-27:
How do I compare the data in the ground truth and the data in the result? Their timestamp formats are somewhat different, and the data difference is relatively large in comparison. how do I process these two sets of data so that they can be compared？
This is the last 200 lines of gt_imu.csv:
---
1520528008679532414,-0.1401063200,-0.0942401146,0.5480597905,0.9982992229,0.0246903362,-0.0509076355,-0.0140521007
1520528008687865414,-0.1409454376,-0.0933471319,0.5474460063,0.9983739118,0.0262188295,-0.0480752843,-0.0158389480
1520528008696199414,-0.1418581051,-0.0930835392,0.5469279191,0.9984973989,0.0250166963,-0.0454655217,-0.0176066941
1520528008704532414,-0.1428648166,-0.0925645206,0.5461513490,0.9985698697,0.0256217837,-0.0427046013,-0.0194436739
1520528008712865414,-0.1434035767,-0.0917786206,0.5451344084,0.9985657033,0.0275662806,-0.0412877256,-0.0200489425
1520528008721198414,-0.1441892367,-0.0916555930,0.5446353217,0.9985242076,0.0282629794,-0.0405108888,-0.0225716355
1520528008729532414,-0.1450975785,-0.0909397843,0.5436407340,0.9985550186,0.0291808143,-0.0378452469,-0.0245782858
1520528008737865414,-0.1459673601,-0.0901280418,0.5430011430,0.9986164486,0.0293279938,-0.0358842072,-0.0248471541
1520528008746198414,-0.1466033062,-0.0894532402,0.5421094496,0.9985601876,0.0319102142,-0.0342996390,-0.0261309163
1520528008754532414,-0.1468364962,-0.0891514651,0.5413950042,0.9985071182,0.0328260884,-0.0332441503,-0.0282985733
1520528008762865414,-0.1474400360,-0.0886640963,0.5407756030,0.9985086397,0.0337055843,-0.0319191347,-0.0287332374
1520528008771199414,-0.1482597092,-0.0881657121,0.5399135433,0.9985578420,0.0346437243,-0.0295083066,-0.0284834762
1520528008779532414,-0.1487377425,-0.0878279681,0.5390189444,0.9985283075,0.0358307183,-0.0289382451,-0.0286348854
1520528008787865414,-0.1491599193,-0.0873402738,0.5381167654,0.9984413471,0.0385933788,-0.0281874251,-0.0288252759
1520528008796198414,-0.1496101955,-0.0869639638,0.5373885167,0.9984276936,0.0396102271,-0.0269148845,-0.0291334772
1520528008804532414,-0.1501502513,-0.0866186691,0.5363978139,0.9983715884,0.0412914682,-0.0253484230,-0.0301105218
1520528008812865414,-0.1507480403,-0.0863472908,0.5359592902,0.9983588766,0.0419920098,-0.0244206139,-0.0303291656
1520528008821199414,-0.1511328855,-0.0860572639,0.5353809381,0.9983080586,0.0427597546,-0.0246750875,-0.0307207341
1520528008829532414,-0.1513903108,-0.0851057612,0.5338858676,0.9982011269,0.0470883390,-0.0235111000,-0.0287128314
1520528008837865414,-0.1516703479,-0.0848905883,0.5324201553,0.9980250175,0.0517442228,-0.0220572516,-0.0279656486
1520528008846198414,-0.1513679394,-0.0848635434,0.5315557598,0.9979831699,0.0521144146,-0.0235772581,-0.0275280443
1520528008854532414,-0.1520797749,-0.0845977734,0.5309803864,0.9979070001,0.0539907080,-0.0220287803,-0.0279527353
1520528008862865414,-0.1522493851,-0.0845610929,0.5302570755,0.9978873144,0.0544970994,-0.0218131183,-0.0278417278
1520528008871198414,-0.1524793130,-0.0847987912,0.5299757885,0.9978260780,0.0550631585,-0.0217537967,-0.0289471756
1520528008879532414,-0.1525737360,-0.0845485949,0.5290364871,0.9976776957,0.0575138996,-0.0221938221,-0.0289620628
1520528008887865414,-0.1525640906,-0.0846214096,0.5286359621,0.9976359377,0.0587811340,-0.0215596300,-0.0283283673
1520528008896199414,-0.1526071366,-0.0846025543,0.5277697345,0.9976632795,0.0591094189,-0.0213619361,-0.0267903893
1520528008904532414,-0.1527669616,-0.0841706475,0.5270575114,0.9976227102,0.0608319705,-0.0205924361,-0.0249870179
1520528008912865414,-0.1528670269,-0.0843294889,0.5269591818,0.9976113271,0.0605259942,-0.0207454421,-0.0260359500
1520528008921198414,-0.1529230033,-0.0842920019,0.5265952909,0.9975911888,0.0611263192,-0.0203369622,-0.0257254954
1520528008929532414,-0.1528147269,-0.0844683510,0.5264688593,0.9975812746,0.0610290169,-0.0198979453,-0.0266670486
1520528008937865414,-0.1529010159,-0.0845106194,0.5261601394,0.9976061018,0.0615728335,-0.0187191822,-0.0253069975
1520528008946199414,-0.1527516099,-0.0841484759,0.5258088132,0.9976379920,0.0611727154,-0.0184112702,-0.0252460072
1520528008954532414,-0.1524739386,-0.0842967615,0.5259644287,0.9976440051,0.0604618263,-0.0184331291,-0.0266650780
1520528008962865414,-0.1525515874,-0.0839827644,0.5258428822,0.9978321553,0.0580536665,-0.0184821583,-0.0248831568
1520528008971198414,-0.1520825315,-0.0844109490,0.5262244757,0.9977907960,0.0571274954,-0.0198661432,-0.0274829598
1520528008979532414,-0.1519085688,-0.0843255741,0.5265491990,0.9979826650,0.0543094857,-0.0196024927,-0.0263973942
1520528008987865414,-0.1517543375,-0.0841889851,0.5267156056,0.9980955948,0.0526553332,-0.0192124705,-0.0257581150
1520528008996198414,-0.1513388512,-0.0843364988,0.5273445674,0.9982954027,0.0489732427,-0.0199552684,-0.0246920568
1520528009004532414,-0.1509887949,-0.0841905037,0.5276193220,0.9983325759,0.0471529354,-0.0213550398,-0.0255466406
1520528009012865414,-0.1506273770,-0.0834796334,0.5276286132,0.9985115959,0.0449871238,-0.0210552155,-0.0225261932
1520528009021199414,-0.1501096187,-0.0836062087,0.5283628210,0.9984669540,0.0443687954,-0.0218732504,-0.0248336988
1520528009029532414,-0.1499141773,-0.0835418058,0.5292432346,0.9987001370,0.0407688524,-0.0205873049,-0.0226296266
1520528009037865414,-0.1495854367,-0.0834622628,0.5299568062,0.9988073673,0.0373665129,-0.0216952659,-0.0227354843
1520528009046198414,-0.1490271825,-0.0834207763,0.5305177443,0.9988647532,0.0348415205,-0.0237768516,-0.0221344653
1520528009054532414,-0.1492075019,-0.0830179500,0.5312679398,0.9990387117,0.0314353924,-0.0224454465,-0.0207284951
1520528009062865414,-0.1486280644,-0.0829927631,0.5317908127,0.9990569790,0.0297248342,-0.0243697626,-0.0201916215
1520528009071199414,-0.1481408439,-0.0831872561,0.5327568534,0.9990335757,0.0290087482,-0.0255531423,-0.0209151620
1520528009079532414,-0.1475928418,-0.0830122148,0.5333564039,0.9989949003,0.0281089213,-0.0276523911,-0.0213171970
1520528009087865414,-0.1470826209,-0.0827452022,0.5336129844,0.9989494041,0.0292440149,-0.0286725077,-0.0205611993
1520528009096198414,-0.1470866019,-0.0826848358,0.5344352848,0.9991912592,0.0246560007,-0.0259108215,-0.0183722205
1520528009104532414,-0.1463774914,-0.0826163558,0.5351364277,0.9991398790,0.0273463258,-0.0257558839,-0.0175589061
1520528009112865414,-0.1458806263,-0.0821212596,0.5356632623,0.9991078437,0.0263305398,-0.0279226906,-0.0176222197
1520528009121198414,-0.1454798367,-0.0817677077,0.5358858230,0.9991175643,0.0260594750,-0.0282096194,-0.0170062858
1520528009129532414,-0.1452183681,-0.0818085753,0.5363866015,0.9991298898,0.0253763380,-0.0285628837,-0.0167232331
1520528009137865414,-0.1445962932,-0.0815442291,0.5369317152,0.9990845841,0.0287895682,-0.0275181875,-0.0156174257
1520528009146199414,-0.1443056719,-0.0812323168,0.5375253449,0.9991437066,0.0266404476,-0.0286654861,-0.0134324241
1520528009154532414,-0.1439494838,-0.0814139439,0.5382485288,0.9991027150,0.0290299080,-0.0276825125,-0.0135907251
1520528009162865414,-0.1436357860,-0.0811483753,0.5386177254,0.9991313683,0.0288165130,-0.0271096332,-0.0130837788
1520528009171198414,-0.1430281438,-0.0812706516,0.5394768504,0.9991619266,0.0272937204,-0.0279128280,-0.0123033008
1520528009179532414,-0.1426529264,-0.0806367134,0.5397602135,0.9991893532,0.0273941799,-0.0278551381,-0.0097101332
1520528009187865414,-0.1419927656,-0.0803074442,0.5402238283,0.9991256115,0.0283128586,-0.0291428736,-0.0098532966
1520528009196199414,-0.1417459258,-0.0799162408,0.5403107505,0.9990703549,0.0302790391,-0.0290455042,-0.0098977028
1520528009204532414,-0.1407920826,-0.0802446086,0.5412698198,0.9990717700,0.0283091514,-0.0314102988,-0.0082209201
1520528009212865414,-0.1405384968,-0.0796097437,0.5416896061,0.9990594963,0.0300073864,-0.0305210162,-0.0069388141
1520528009221198414,-0.1398705763,-0.0795047772,0.5421359855,0.9990820243,0.0285518428,-0.0315024840,-0.0052435205
1520528009229532414,-0.1391890534,-0.0785191805,0.5420901021,0.9990896653,0.0297705781,-0.0305156337,-0.0015328229
[... 312 lines cut ...]
1520528009.16506 -0.0509153259119268  0.0884315056444326 -0.0663892954545823 -0.00645898706597481 0.00841188549192228 0.00336325613700642 0.999938103172746
1520528009.21506 -0.0558727420100482  0.0858937087735925 -0.0697104626138843 -0.0072882367881853 0.000279077686515795 0.00433075555839258 0.999964023491074
1520528009.26507 -0.0622714922994058  0.0831717951421595 -0.0741239633514288 -0.00173258384286967 -0.0128510619803884 0.0021328605041941 0.999913645904122
[... cut ...]
1520528010.16509 -0.0387510487025806   0.160328142136707 -0.0852522170949735 -0.000348101194948047 -0.0196707691413904 -0.00628448525992132 0.999786699707474
1520528010.2151 -0.0367704907196186   0.163629360499796 -0.0843118734885974 -0.00243337714354758 -0.019742763862147 -0.00745143405266875 0.999774363584764
1520528010.2651 -0.0357817190464075   0.166836495896622 -0.0840472518780749 -0.00151573010503603 -0.0213219801338921 -0.00842596357384806 0.999736004584846
1520528010.3151 -0.0358134361470846   0.168455278492694 -0.0857483148138746 0.00181733058704359 -0.0266066907965064 -0.0105247524839307 0.999588920956785
///

lukasvst, 2023-08-05:
You can have a look at the Python evaluation tools: https://github.com/lukasvst/dm-vio-python-tools
There's examples how to evaluate the trajectory on EuRoC, TUM-VI and 4Seasons, and if you are interested in the details you can check-out the code.

GY56723-debug, 2023-08-07:
Thanks a lot for your reply. I will close the Issue

closed 2023-08-07 by GY56723-debug