lukasvst/dm-vio #50: gtsam symbol error

Neoyning, 2023-11-17:
Hi there, after running the executable dmvio_dataset, I got the following error.  GTSAM 4.2.0 is installed in a specific path and linked to the project as follows,

find_package(GTSAM 4.2.0 EXACT REQUIRED PATHS "/home/neoy/alter_lib" NO_DEFAULT_PATH)

./dmvio_dataset
./dmvio_dataset: Symbol `_ZTVN5gtsam6FactorE' has different size in shared object, consider re-linking
./dmvio_dataset: Symbol `_ZTVN5gtsam16NoiseModelFactorE' has different size in shared object, consider re-linking
./dmvio_dataset: Symbol `_ZTVN5gtsam13HessianFactorE' has different size in shared object, consider re-linking
./dmvio_dataset: Symbol `_ZTVN5gtsam20NonlinearFactorGraphE' has different size in shared object, consider re-linking
./dmvio_dataset: Symbol `_ZTVN5gtsam14JacobianFactorE' has different size in shared object, consider re-linking
./dmvio_dataset: Symbol `_ZTVN5gtsam9ImuFactorE' has different size in shared object, consider re-linking
./dmvio_dataset: symbol lookup error: ./dmvio_dataset: undefined symbol: _ZN5gtsam14JacobianFactorC1EmRKN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEERKNS2_IdLin1ELi1ELi0ELin1ELi1EEERKN5boost10shared_ptrINS_10noiseModel8DiagonalEEE

Does anyone know how to fix it? Thanks!

lukasvst, 2023-11-19:
You can try deleting the build folders for both DM-VIO and GTSAM, and rebuild them. (Also run make sure to run `sudo make install` in the gtsam build dir after building and before building DM-VIO).
Otherwise, make sure that you have only one version of GTSAM installed on your machine.

Neoyning, 2023-11-20:
Thanks! By keeping just one version of GTSAM, the executable is properly running~

I installed two versions of GTSAM (4.3.0 and 4.2.0). GTSAM 4.3.0 was installed by 'sudo make install' with the default installation folder "/usr/local/..", GTSAM 4.2.0 was installed by 'make install' in my specified folder. 

After I uninstall GTSAM 4.3.0 by 'sudo xargs rm < install_manifest.txt' in its build folder. And the add the "lib folder of the specified install folder" to dynamic library path($LD_LIBRARY_PATH) .  The executable dmvio_dataset now can properly run in TUM-VI dataset.

Thanks again for your help!

closed 2023-11-20 by Neoyning