lukasvst/dm-vio #66: Path Problem

Yasintrkmn, 2025-01-03:
I try to run dm-vio on my own dataset. 
I created Inerpolated IMU, timestamp and necessary yaml files.
When I run `run_dmvio.py` script, I get following:

`ImageFolderReader: got 1200 files in ./cam0!
IMU Id: ...
Segmentation fault (core dumped)
Executing: cp .../result.txt ...txt
cp: cannot stat '.../result.txt': No such file or directory
Executing: cp .../resultKFs.txt ....txt`

How to solve the problem?

Would be better more clear explanations about running it on private datasets.
Thanks for your work.

Yasintrkmn, 2025-01-03:
> I try to run dm-vio on my own dataset. I created Inerpolated IMU, timestamp and necessary yaml files. When I run `run_dmvio.py` script, I get following:
>
> `ImageFolderReader: got 1200 files in [...]

> What is exposures?
> How to solve the problem?
> 
> Would be better more clear explanations about running it on private datasets. Thanks for your work.

closed 2025-01-06 by Yasintrkmn

nushakrishnan, 2025-02-16:
Hello, I am facing the same "Assertion '0' failed" error when running DM-VIO on my custom data. Can you share how you resolved the issue? Resizing the images does not work for me unfortunately.